OpenHantek/openhantek #333: Does not appear attenuation

JefDrones, 2021-12-21:
I installed it after many attempts on Ubuntu, but the checkbox x1 x10 doesn't appear, how can I change the interface?
Thanks guys

Ho-Ro, 2021-12-21:
> but the checkbox x1 x10 doesn't appear

Which checkbox? - openhantek does not have a x1 x10 checkbox (neither does Openhantek6022). 
Which kind of HW do you use? In case it's a Hantek 6022 I would propose to check [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022), because openhantek is [unmaintained](#277). It also provides the [latest binaries](https://github.com/OpenHantek/OpenHantek6022/releases) for almost all major OS (Linux (also for RPi), MacOS, FreeBSD as well as Windows) - ready to install easily with your package manager - in your case a simple download followed by `apt install ./openhantek_..._amd64.deb` will do the magic, it even installs the needed dependencies.

JefDrones, 2021-12-21:
[image]
My HW is 6022BE, I installed OpenHantek6022, sorry I said wrong in the last post.
See the screen print, the option to choose the attenuation does not appear

Ho-Ro, 2021-12-21:
Why did you install this (very outdated) version?
Where did you get it? 
Why don't you use the the [latest release](https://github.com/OpenHantek/OpenHantek6022/releases/download/3.2.5/openhantek_20211216-31c5ee2_amd64.deb) (built under Ubuntu 20.04)?
In case you have an older Ubuntu install, you can use previous [Release 3.2.4](https://github.com/OpenHantek/OpenHantek6022/releases/download/3.2.4/openhantek_20210714-7483fb2_amd64.deb) that was built under Ubuntu 18.04.
If you still have issues, please report them in the [correct repo](https://github.com/OpenHantek/OpenHantek6022).

JefDrones, 2021-12-21:
Why did you install this (very outdated) version?
Where did you get it?

Oh my God, I didn't know, I think I got it from the openhantek.org site, I struggled so hard to install it because it's been 4 days since I started using Linux, I've downloaded so much that I don't even know anymore.

Why don't you use the latest release (built under Ubuntu 20.04)?
In case you have an older Ubuntu install, you can use a previous Release 3.2.4 that was built under Ubuntu 18.04.
If you still have issues, please report them in the correct repo.

I will look here for the latest version as my Ubuntu is 20.04