OpenHantek/openhantek #337: Can't connect DSO2D15

piudeferro, 2023-01-06:
I have installed OpenHantek following the steps from GitHub, libraries and so on, but when I open the program it cannot connect to the oscilloscope because it is not one of the ones listed in the program. I have tried the unsupported tab and it does detect it but it returns the error that it is not implemented.
[image]
[image]
[image]
[image]

Ho-Ro, 2023-01-06:
Please check #277!

piudeferro, 2023-01-06:
Do you mean that there is no support and since there is no maintainer, the case related to openhantek is closed?

Ho-Ro, 2023-01-07:
Who should work on this issue? You could do what I did. Some years ago I bought a Hantek 6022BE because it had linux support by openhantek. When the maintainer stopped his support I forked my project [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022), removed all other HW related code (because I cannot develop for HW that is not available at my desk) and implemented the changes I wanted to have.

piudeferro, 2023-01-07:
Could I ask you for a little help? I've never programmed or developed code or implemented, but I don't think it's very difficult either. A short time ago I started arduino projects with my son, and I have read that there are programs that modify windows dll's to be used in devices that do not have linux drivers. If I could do it, once the path was done, I would repeat it for the power supply and for the bench multimeter. I have done tests with Bottles and with Wine, and I almost had to reinstall openSUSE because it had so many changes that I couldn't start the pc.
I've been using linux for years but I've never developed or implemented anything, so with a little bit of your help showing me the way I think I'll get it.
Still, thank you very much for your attention.

prash85, 2023-01-07:
* This is not a professional advice but just a comment from a user. 
*
Try connecting USB of your device "AFTER" starting the application for auto detection. I've had issue getting my 6022BE deteccted on windows machine but after a few attempts of "plugging and unplugging" the applications was able to auto detect my device. 
Hope this works for you.

Ho-Ro, 2023-01-07:
> I have read that there are programs that modify windows dll's to be used in devices that do not have linux drivers.

There are two types of windows _driver_, the 1st type just associates an USB device with a low level driver, that allows the HW access while the 2nd type implements the API and does the real communication. On a Linux machine you typically allow the access with `udev` rules, do the low level communication with `libusb` and implement the API in the application.
As a starting point for your development, I could recommend looking at sigrok to see if they already have efforts for support for this device. Otherwise you have to decode the protocol yourself. For the general procedure, I can recommend a very informative video in which (although for a completely different HW) the USB communication is analysed step by step: https://www.youtube.com/watch?v=xJ0Mfuj0KUQ

closed 2023-01-14 by piudeferro