OpenHantek/openhantek #338: Default screenshot location is C:\Windows\System32

misch2, 2023-03-25:
I'm running OpenHantek6022 version 3.3.2.2 on Windows 11.
I downloaded and extracted it into a folder under my profile:
[image]

When I run it and try to produce a screenshot or a hardcopy, it tries to save the result into the Windows `system32` subfolder, as confirmed by process monitor:
[image]

This action fails due to permissions, but OpenHantek doesn't indicate the screenshot failure in any way.

My suggestions are:
 - If any file operation fails (Export, Screenshot, ...), it would be nice to display a warning message.
 - It would be great to save the screenshots either into a subfolder in `%APPDATA%\OpenHantek`, or into the standard windows "Pictures" folder in user profile.

closed 2023-03-26 by misch2

Ho-Ro, 2023-03-26:
How did you solve this issue?

misch2, 2023-03-27:
@Ho-Ro My apologise, this was a user error, not caused by the software. 

OpenHantek (Windows version) apparently saves the screenshots into the folder it has been started from, i.e. not into a fixed location (as would be for example user's profile directory or the "Pictures" subfolder). 

Therefore when I start the OpenHantek by clicking the program icon in the folder where it's stored, screenshots work OK. 

But I was running it by using windows default search function:
[image]
and in this case Windows apparently (?) for some reason use the system folder as the base one. So the OpenHantek just tried to save the screenshots into the current folder, which due to the way I started the SW, was the `c:\windows\system32`.

Ho-Ro, 2023-03-27:
I just thought about this reason. I had no access to a win system during last week, so I couldn't check.  Maybe a possibility could be to create a link where you can define the default location. Another option is to use the menu entry `Save screenshot as..` where you can define the file name as well as the path.
Nevertheless I will check the possibility to raise a warning if the screenshot cannot be saved.

Ho-Ro, 2023-03-27:
@misch2 You posted in the wrong project - this is the old and [unmaintained](#277) predecessor. Nevertheless, I added a rough error message window to OpenHantek6022, that is shown when the export (screenshot & hardcopy as well as CSV & JSON) fails, e.g. due to missing rights or write protection), available in latest [unstable](https://github.com/OpenHantek/OpenHantek6022/releases/tag/unstable) `3.3.2.2-8-g2ea8261`.

[image]