OpenHantek/openhantek #341: Demo mode does not display any traces

softfoot, 2023-11-03:
I just downloaded and installed openhantek_3.3.3-1_win_x64.zip but when I started it "Demo" mode showed all the markers, triger markers, etc etc but did not display any traces.

An earlier release (3.3.2.xx I think) worked OK - I also tried it with the real scope attached - it detected it OK as a DSO-6022BE but as before the traces are missing.

[image]

I'm running under Win7 64 Pro on a laptop.
Regards,
Dave

softfoot, 2024-01-09:
Thanks that fixed it :-)
Dave

Ho-Ro, 2024-01-09:
Hi Dave, which build did you use? MSVC? Please try also the other build and report if the above issue is still active.
Reason could be that the two setups use different OpenGL modes, at least when I d a quick check on my linux system with the windows emulation `wine`.

BTW: You are reporting this issue in the wrong repo - this here is the old [unmaintained](https://github.com/OpenHantek/openhantek/issues/277) predecessor project `openhantek`, while you're talking about [OpenHantek6022](https://github.com/OpenHantek/OpenHantek6022).
Please close your issue here, I will handle this in your [issue #388](https://github.com/OpenHantek/OpenHantek6022/issues/388) on OpenHantek6022.